jackal/jackal #23: Add and test multirobot support

rgariepy, 2015-01-05:
Seems to work in sim here: https://github.com/jackal/jackal/tree/multirobot

Further testing required

mikepurvis, 2015-01-14:
Removing leading slashes on things is a no-brainer. However, before we consider merging the explicit tf_prefix support, I'd like to at least build a toy prototype of doing it the other way (tf proxying), and engage the community regarding the preferred approach.

If we do end up going this route, it would probably make the most sense to abolish the control.yaml file and bring its contents into the launch file. Then we can use the new `<rosparam subst_value="true">` attribute to get inline substitutions in the yaml rather than having separate `<param>` tags for everything referencing a frame.

civerachb-cpr, 2020-06-02:
Closing this as it's super old and the multi-robot work is being done elsewhere now.

closed 2020-06-02 by civerachb-cpr