jackal/jackal #24: Localization config needs updating

mikepurvis, 2015-01-15:
Getting the following errors on startup:

```
[ WARN] [1421336494.127396465]: Configuration vector for odom0_config should have 15 entries.
[ WARN] [1421336494.157052842]: Detected deprecated parameter remove_gravitational_acceleration. Please specify this parameter for each IMU, e.g., imu0_remove_gravitational_acceleration
[ WARN] [1421336494.158280263]: Configuration vector for imu0_config should have 15 entries.
```

The section in question is here: https://github.com/jackal/jackal/blob/indigo-devel/jackal_control/config/control.yaml#L37

@ayrton04 can confirm, but it looks like the [new state members are for acceleration](https://github.com/cra-ros-pkg/robot_localization/blob/master/include/robot_localization/filter_common.h#L40). I guess we probably _do_ want this from the IMU, and _don't_ want it from the wheel odometry.

ayrton04, 2015-01-16:
Correct!

tonybaltovski, 2016-05-11:
Done in 3be4068b5b5cac1c08b97ecc020c19f6937e68ea.

closed 2016-05-11 by tonybaltovski