jackal/jackal #41: Bad Documentation

MengGuo, 2016-09-24:
As a official repository for a pricy commercial product, I found the documentation very unsatisfactory. 

Any brief explanation about the code structure would be helpful, i.e., what does what, how it is related to [Jackal_desktop](https://github.com/jackal/jackal_desktop) and [jackal_robot](https://github.com/jackal/jackal_robot).

I know the potential users are promising engineers but that doesn't mean they don't need a smooth start.

tonybaltovski, 2016-09-24:
Have you take a look at the [ROS wiki](http://wiki.ros.org/Robots/Jackal)? ie [jackal_viz](http://wiki.ros.org/jackal_viz?distro=indigo) and [jackal_robot](http://wiki.ros.org/jackal_robot).  What do you find missing to ensure a smooth start?

MengGuo, 2016-09-25:
Yes, the ROS wiki and the tutorial is the first thing I opened. There are three things we tried the first week when we are working on the Jackal: (1) connected to WiFi;  (2) control it by manually send a Twist message over ROS from another computer so it move; (3) get data from the Velodyne. 

(1) was solved eventually but the default bridge setting is confusing.  But (2) is still under going as nothing is mentioned in the tutorial. Joystick seems to be focus but IMHO it's useless for research; (3)  run the visualization viz and everything is easy to follow, but again we are more interested in the raw data and how to turn it on/off, how to change it scanning frequency. 

They are just my experience. I send a question about (2) through your official website. If you prefer, I can post my questions here so more people can see.

ibaranov-cp, 2016-09-26:
Don't forget the general ROS guide we have, which as a first step teaches how to manually send Twist messages, and echoing raw data:

https://www.clearpathrobotics.com/assets/guides/ros/

Let us know where we can add to the guide :)

tonybaltovski, 2016-09-26:
Can you clarify what confused you on (1)?  The bridged Ethernet ports are totally separate then the wireless.  For (2), the ROS standard is move a robot by sending a Twist message to the topic `cmd_vel` which can be done using a simple Twist publisher as pointed out above. Lastly, all the information for the Velodyne can be found on the ROS Wiki [here](http://wiki.ros.org/velodyne_driver?distro=indigo).

This is not the ideal place to post questions as it is an issue tracker for the code contained in this repository.  A better option would be to contact customer support or on ROS Answers if you want more people to see.

tonybaltovski, 2016-10-14:
Any updates on this @MengGuo ?

MengGuo, 2016-10-14:
Hi Tony, we get the IP sorted out. Thanks! But sometimes after we start the Jackal, WiFi doesn't connect automatically. Maybe it is a Ubuntu thing. Also have you met the case that Jackal is not connected to the Internet by wireless, even though the LAN is fully connected? We are now always connecting Jackal by wire to the Internet.

tonybaltovski, 2016-10-14:
I can't recall a Jackal not automatically connecting to wireless.  It should automatically connect to the wireless and have Internet.  Did you modify anything related to the Networking configuration?

MengGuo, 2016-10-14:
We set in `wcid` to choose a static IP. Don't think that's the reason though.  

I will paste the full `ifconfig` here next week for you to take a look.

amiyajima, 2016-10-17:
Here are the results of running `ifconfig`

administrator@Clearpath1:~/catkin_ws/src/duk03_jackal/launch$ ifconfig
br0       Link encap:Ethernet  HWaddr 00:0b:ab:ad:60:8e  
          inet addr:192.168.1.11  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::20b:abff:fead:608e/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:57153 errors:0 dropped:0 overruns:0 frame:0
          TX packets:46 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:63268990 (63.2 MB)  TX bytes:9553 (9.5 KB)

eth0      Link encap:Ethernet  HWaddr 00:0b:ab:ad:60:8f  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Memory:f7c00000-f7c1ffff 

eth1      Link encap:Ethernet  HWaddr 00:0b:ab:ad:60:8e  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:57153 errors:0 dropped:0 overruns:0 frame:0
          TX packets:46 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:64297744 (64.2 MB)  TX bytes:9749 (9.7 KB)
          Interrupt:20 Memory:f7e00000-f7e20000 

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:75526 errors:0 dropped:0 overruns:0 frame:0
          TX packets:75526 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:18284855 (18.2 MB)  TX bytes:18284855 (18.2 MB)

wlan0     Link encap:Ethernet  HWaddr 7c:5c:f8:25:bf:5c  
          inet addr:10.0.0.4  Bcast:10.255.255.255  Mask:255.0.0.0
          inet6 addr: fe80::7e5c:f8ff:fe25:bf5c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:228 errors:0 dropped:0 overruns:0 frame:0
          TX packets:189 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:22398 (22.3 KB)  TX bytes:35170 (35.1 KB)

mcoteCPR, 2016-10-18:
Hello! 

I think this would be best handled through our support tickets rather than Github, as this doesn't seem to be an issue related to the actual code base.

Feel free to make a ticket and I'll be happy to help you there! ;)

MengGuo, 2016-10-18:
Done, thanks!

closed 2016-10-19 by tonybaltovski