jackal/jackal #42: Rviz Error: No Map Received

amiyajima, 2016-10-06:
We ran `roslaunch jackal_navigation gmapping_demo.launch` but in rviz we see the point cloud but no map is produced. Under the map tab we see an error saying "No Map Received". 

Is this an issue with gmapping configurations?

tonybaltovski, 2016-10-06:
The gmapping demo is configured for a laser scanner.  I assume you are not using one since you are seeing point clouds?  You can use [pointcloud_to_laserscan](http://wiki.ros.org/pointcloud_to_laserscan) to convert from3D Point Cloud into a 2D laser scan for gmapping.

MengGuo, 2016-10-06:
We have Velodyne on board, which is a laser scanner, right? the visualization in Rivz is subscribed to point cloud though. How should we configure `gmapping` with `Velodyne` then?

tonybaltovski, 2016-10-07:
Create a launch file to use `pointcloud_to_laserscan` such as the one below and add it to the upstart job.  Make sure `pointcloud_to_laserscan` is installed.

```
<node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">

    <remap from="cloud_in" to="/velodyne_points"/>
    <remap from="scan" to="/front/scan"/>
    <rosparam>
        transform_tolerance: 0.01
        min_height: 0.25
        max_height: 0.75

        angle_min: -1.5708
        angle_max: 1.5708
        angle_increment: 0.087
        scan_time: 0.1
        range_min: 0.9
        range_max: 130
        use_inf: true
        concurrency_level: 0
    </rosparam>

</node>
```

MengGuo, 2016-10-07:
Thanks! where is the upstart file normally located in Jackal?

tonybaltovski, 2016-10-07:
No worries.  Make your launch file and a good idea would be to test it.  Once you verified it works, you can add it to upstart job using `rosrun robot_upstart install my_custom_package/launch --job ros --augment`.

MengGuo, 2016-10-07:
Got it. Is that permanent or it needs to be done each time I turn Jackal on?

tonybaltovski, 2016-10-07:
It will be permanent when the upstart job is augmented.

MengGuo, 2016-10-12:
It worked, Tony. Thanks!

tonybaltovski, 2016-10-14:
May I close this issue?

MengGuo, 2016-10-14:
Yep, thanks!

closed 2016-10-14 by tonybaltovski

syqssyy, 2022-03-09:
i