jackal/jackal #43: Rviz error for amcl_demo

MengGuo, 2016-10-12:
Hello Tony, 

we have reached the last part of the tutorial. We generate a .yaml map using gmapping and installed amcl on Jackal. 

When we launched the amcl_demo.launch, Jackal is not showing up in Rviz see the figure below:

[image]

any idea what might be the reason? Thanks,
/Meng

tonybaltovski, 2016-10-14:
Looking at image you posted, getting data to the desktop machine.  Can you try echoing the tf topic using `rostopic echo /tf` on the desktop machine?

MengGuo, 2016-10-14:
Hi Tony, the error disappeared after we restart the Jackal. We think it is a network issue. 

Also, for other people's reference, the `amcl_demo`also needs the`pointcloud_to_laserscan` since we have Velodyne.

closed 2016-10-14 by tonybaltovski