jackal/jackal #44: Jackal does not avoid obstacles

amiyajima, 2016-10-17:
When running the odometry navigation demo, our jackal does not avoid static or dynamic obstacles. Is there an issue with our configuration for the Velodyne lidar?

MengGuo, 2016-10-17:
Also the same thing happens with gmapping and amcl demo, since they use the same `move_base.launch`.

MengGuo, 2016-10-18:
Also where is `costmap_common_params.yaml` located? I guess that's where you define `point_cloud_sensor` as one of the sensors that is going to be passing information to the cost map?

MengGuo, 2016-10-18:
Is it because in `jackal/jackal_navigation/params/costmap_common_params.yaml` it didn't list `laser_scan_sensor` as one of sensors that is going to be passing information to the cost map?

How do we configure it for `Velodyne LiDAR` then?

mcoteCPR, 2016-10-18:
Hey Guys

Once again this questions seems to be more of a specific robot issue, would you mind filling out a support ticket [here](https://support.clearpathrobotics.com/hc/en-us/requests/new?ticket_form_id=59673). I'll be happy to help you sort this out there!

MengGuo, 2016-10-18:
This I don't agree. Clearly, it is an issue related to the configuration file of `move_base`, which should be very beneficial for other users in case of similar situations. 

So I think it is better to tackle it here. Also I want to refer my colleague here regarding what's the issue. Thank you!

mcoteCPR, 2016-10-18:
Hi Meng,

This seems to be an issue with your particular sensor set up rather than move_base, which is why we believe this is more robot specific issue.

Are you still using the point cloud to laser scan package? It's likely your height is not properly set, so the Jackal is not avoiding objects that are too low. I'd encourage you to try changing the parameter for this package and see if it helps, specifically the height parameters.

MengGuo, 2016-10-18:
Thanks for understanding, Martin. 

Yes, we are converting the pointcould to 2Dlaser, by which the gmapping and amcl are working fine. 

I was standing and blocking completely the front of the lidar,  so I don't think it's because of the height setting. 

Can you help me to understand why `jackal/jackal_navigation/params/costmap_common_params.yaml` does not include `point_cloud_sensor`? or after we transform pointclould to 2Dlaser, `move_base` can take that automatically even if it is not specified in the `costmap_common_params.yaml`?

tonybaltovski, 2016-10-19:
Meng,

Are you able to visualise the scan data?  If so, do you see obstacles at the level of the Jackal?

`jackal/jackal_navigation/params/costmap_common_params.yaml` can use a point cloud sensor as shown [here](http://wiki.ros.org/costmap_2d/hydro/obstacles).  `gmapping` and `amcl` only can use a laser scan. Therefore, the general use would be to use a laser scan for the costmaps.

MengGuo, 2016-10-19:
Yea, I can visualize the `pointcloud` from Velodyne in Rviz by subscripting to the topic `/pointcolud2`. And when I stand in front of Jackal, the pointcloud reflects that, also all other obstacles in the environment. 

So you don't think that not including point cloud sensor in `costmap_common_params.yaml` is the reason? We also tried to run `odom_navigation_demo.launch` with `pointcloud_to_laserscan`running at the same time. Jackal can't avoid obstacle either.

tonybaltovski, 2016-10-19:
How about form the `/front/scan` topic as a laser scan?

I don't believe that is the case since laser data is provided.  Can you send a bag of what you are trying?

amiyajima, 2016-10-20:
The outputs at `/front/scan` do not seem to change when an obstacle is present.

Here is a bag of `/front/scan`. We dynamically introduced an obstacle while we were recording the bag but it didn't look like the output was changing.
[front-scan-output.bag.tar.gz](https://github.com/jackal/jackal/files/542644/front-scan-output.bag.tar.gz)

amiyajima, 2016-10-20:
Here is a video that shows the point cloud and laser scan from `/front/scan`.

The laser scan does not detect nearby obstacles, but does detect obstacles that are a couple of feet away. However, this seems to not be used in colission avoidance as we still have not gotten that to work.

https://youtu.be/tizxYetILUk

tonybaltovski, 2016-10-21:
Looking at your bag, the actual laser points seem rather sparse 36 points for Pi range.  Can you post your `pointclound_to_laser` configuration?

MengGuo, 2016-10-21:
Hi Tony, we are using the one you suggested in [another issue](https://github.com/jackal/jackal/issues/42) 

```
<node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">

    <remap from="cloud_in" to="/velodyne_points"/>
    <remap from="scan" to="/front/scan"/>
    <rosparam>
        transform_tolerance: 0.01
        min_height: 0.25
        max_height: 0.75

        angle_min: -1.5708
        angle_max: 1.5708
        angle_increment: 0.087
        scan_time: 0.1
        range_min: 0.9
        range_max: 130
        use_inf: true
        concurrency_level: 0
    </rosparam>

</node>
```

MengGuo, 2016-10-24:
Hi Tony, have you got time to take a look at this? the `angle_increment=0.087`, which gives around `36` samples. So that's correct. 

If you take a look at the video, the `laserscan` is recorded, which however is not used in the `move_base` module.

mcoteCPR, 2016-10-24:
Hi Meng,

Tony is working on a more comprehensive solution for you using a voxel layer with the full pointcloud. We're doing some tests on our in house unit, and will tie everything into an enviroment variable when we're done. We can't promise a definite timeline, but we'll let you know when it's ready, and it should fix your obstacles avoidance issues.

In the mean time, try decreasing to `angle_increment=0.004`` to improve the amount of points per scan.        As well,  angle_min: -3.14 and angle_max: 3.14

MengGuo, 2016-10-24:
Get it, Martin. Thanks! 

We will try that first.

MengGuo, 2016-10-27:
Hello Martain and Tony,

We finally solve the issue by adding **a static tf transformation from `velodyne` to `front_laser`**. The warning message in `odom_navigation_demo.launch` says *there the message from [/odom /front_laser] 100% dropped*. 

Then I checked that even though `/font/scan` is publishing, there is no `/front_laser` in the `tf` tree, only from `/odom` to `/velodyne`. So I added the following node to pointcould_to_laserscan launch:

> `<node pkg="tf" type="static_transform_publisher" name="velodyne_to_front_laser" args="0 0 0 0 0 0 /velodyne /font_laser 100" />`

Now the local and global planners in `move_base` take care of obstacles, see our [first try](https://vimeo.com/189086502).

closed 2018-03-12 by dniewinski