jackal/jackal #46: Kinetic?

liborw, 2017-01-20:
Has anyone tried to use jackal with kinetic?

TheDash, 2017-06-19:
@liborw We are planning a July-August release.

rarmstrong22, 2017-08-18:
Any word on the Kinetic release? I see there's a kinetic-devel branch.

dniewinski, 2018-03-12:
Sorry for the delay on this.  Kinetic is currently in development and we are are hoping to release it within the next month or two.  This issue will be updated when that is completed.

rgreid, 2018-05-01:
Hi guys, any update on the Kinetic release? Thanks.  @dniewinski @ibaranov-cp

dniewinski, 2018-06-09:
Despite many delays, we have finally finished and released support for Kinetic! Many of the packages should be available through http://packages.ros.org or from http://packages.clearpathrobotics.com

closed 2018-06-09 by dniewinski