jackal/jackal #53: Error when run odom_navigation_demo.launch

wael-abdin, 2018-01-04:
I followed the installation process for installing Clearpath’s Jackal simulation, desktop, and navigation packages as explained in the following link ;
`http://www.clearpathrobotics.com/assets/guides/ros/ROS%20Navigation%20Basics.html` 

hen i tried to run the following code ;
`roslaunch jackal_navigation odom_navigation_demo.launch`

the following error appears;
`core service [/rosout] found
ERROR: cannot launch node of type [move_base/move_base]: move_base
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/wael/jackal_navigation/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete`

i used os ; ubuntu 14.04 

the following is the log file appears after run `roslaunch jackal_navigation odom_navigation_demo.launch`

`... logging to /home/wael/.ros/log/84bd5cd8-f130-11e7-8639-606720841920/roslaunch-wael-Inspiron-N5010-5073.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wael-Inspiron-N5010:40262/

SUMMARY
========

PARAMETERS
 * /move_base/TrajectoryPlannerROS/acc_lim_theta: 20.0
 * /move_base/TrajectoryPlannerROS/acc_lim_x: 10.0
 * /move_base/TrajectoryPlannerROS/angular_sim_granularity: 0.02
 * /move_base/TrajectoryPlannerROS/controller_frequency: 20.0
 * /move_base/TrajectoryPlannerROS/dwa: True
 * /move_base/TrajectoryPlannerROS/escape_reset_dist: 0.1
 * /move_base/TrajectoryPlannerROS/escape_reset_theta: 0.1
 * /move_base/TrajectoryPlannerROS/escape_vel: -0.5
 * /move_base/TrajectoryPlannerROS/gdist_scale: 1.0
 * /move_base/TrajectoryPlannerROS/heading_lookahead: 0.325
 * /move_base/TrajectoryPlannerROS/heading_scoring: False
 * /move_base/TrajectoryPlannerROS/heading_scoring_timestep: 0.8
 * /move_base/TrajectoryPlannerROS/holonomic_robot: False
 * /move_base/TrajectoryPlannerROS/latch_xy_goal_tolerance: False
 * /move_base/TrajectoryPlannerROS/max_vel_theta: 1.57
 * /move_base/TrajectoryPlannerROS/max_vel_x: 0.5
 * /move_base/TrajectoryPlannerROS/meter_scoring: True
 * /move_base/TrajectoryPlannerROS/min_in_place_vel_theta: 0.314
 * /move_base/TrajectoryPlannerROS/min_vel_theta: -1.57
 * /move_base/TrajectoryPlannerROS/min_vel_x: 0.1
 * /move_base/TrajectoryPlannerROS/occdist_scale: 0.1
 * /move_base/TrajectoryPlannerROS/oscillation_reset_dist: 0.05
 * /move_base/TrajectoryPlannerROS/pdist_scale: 0.75
 * /move_base/TrajectoryPlannerROS/publish_cost_grid_pc: True
 * /move_base/TrajectoryPlannerROS/sim_granularity: 0.02
 * /move_base/TrajectoryPlannerROS/sim_time: 2.0
 * /move_base/TrajectoryPlannerROS/simple_attractor: False
 * /move_base/TrajectoryPlannerROS/vtheta_samples: 20
 * /move_base/TrajectoryPlannerROS/vx_samples: 6
 * /move_base/TrajectoryPlannerROS/xy_goal_tolerance: 0.25
 * /move_base/TrajectoryPlannerROS/yaw_goal_tolerance: 0.157
 * /move_base/base_global_planner: navfn/NavfnROS
 * /move_base/base_local_planner: base_local_planne...
 * /move_base/clearing_rotation_allowed: True
 * /move_base/controller_frequency: 20.0
 * /move_base/controller_patience: 15.0
 * /move_base/global_costmap/footprint: [[-0.21, -0.165],...
 * /move_base/global_costmap/footprint_padding: 0.1
 * /move_base/global_costmap/global_frame: odom
 * /move_base/global_costmap/height: 40.0
 * /move_base/global_costmap/inflater_layer/inflation_radius: 0.3
 * /move_base/global_costmap/map_type: costmap
 * /move_base/global_costmap/meter_scoring: True
 * /move_base/global_costmap/obstacle_range: 2.5
 * /move_base/global_costmap/obstacles_layer/observation_sources: scan
 * /move_base/global_costmap/obstacles_layer/scan/clearing: True
 * /move_base/global_costmap/obstacles_layer/scan/data_type: LaserScan
 * /move_base/global_costmap/obstacles_layer/scan/marking: True
 * /move_base/global_costmap/obstacles_layer/scan/max_obstacle_height: 2.0
 * /move_base/global_costmap/obstacles_layer/scan/min_obstacle_height: -2.0
 * /move_base/global_costmap/obstacles_layer/scan/obstacle_range: 2.5
 * /move_base/global_costmap/obstacles_layer/scan/raytrace_range: 3.0
 * /move_base/global_costmap/obstacles_layer/scan/sensor_frame: front_laser
 * /move_base/global_costmap/obstacles_layer/scan/topic: front/scan
 * /move_base/global_costmap/origin_x: -20.0
 * /move_base/global_costmap/origin_y: -20.0
 * /move_base/global_costmap/origin_z: 0.0
 * /move_base/global_costmap/plugins: [{'type': 'costma...
 * /move_base/global_costmap/publish_frequency: 5
 * /move_base/global_costmap/publish_voxel_map: False
 * /move_base/global_costmap/raytrace_range: 3.0
 * /move_base/global_costmap/resolution: 0.05
 * /move_base/global_costmap/robot_base_frame: base_link
 * /move_base/global_costmap/rolling_window: False
 * /move_base/global_costmap/static_map: True
 * /move_base/global_costmap/transform_tolerance: 0.5
 * /move_base/global_costmap/update_frequency: 20
 * /move_base/global_costmap/width: 40.0
 * /move_base/global_costmap/z_resolution: 1
 * /move_base/global_costmap/z_voxels: 2
 * /move_base/local_costmap/footprint: [[-0.21, -0.165],...
 * /move_base/local_costmap/footprint_padding: 0.1
 * /move_base/local_costmap/global_frame: odom
 * /move_base/local_costmap/height: 10.0
 * /move_base/local_costmap/inflater_layer/inflation_radius: 0.3
 * /move_base/local_costmap/map_type: costmap
 * /move_base/local_costmap/meter_scoring: True
 * /move_base/local_costmap/obstacle_range: 2.5
 * /move_base/local_costmap/obstacles_layer/observation_sources: scan
 * /move_base/local_costmap/obstacles_layer/scan/clearing: True
 * /move_base/local_costmap/obstacles_layer/scan/data_type: LaserScan
 * /move_base/local_costmap/obstacles_layer/scan/marking: True
 * /move_base/local_costmap/obstacles_layer/scan/max_obstacle_height: 2.0
 * /move_base/local_costmap/obstacles_layer/scan/min_obstacle_height: -2.0
 * /move_base/local_costmap/obstacles_layer/scan/obstacle_range: 2.5
 * /move_base/local_costmap/obstacles_layer/scan/raytrace_range: 3.0
 * /move_base/local_costmap/obstacles_layer/scan/sensor_frame: front_laser
 * /move_base/local_costmap/obstacles_layer/scan/topic: front/scan
 * /move_base/local_costmap/origin_z: 0.0
 * /move_base/local_costmap/plugins: [{'type': 'costma...
 * /move_base/local_costmap/publish_frequency: 5.0
 * /move_base/local_costmap/publish_voxel_map: False
 * /move_base/local_costmap/raytrace_range: 3.0
 * /move_base/local_costmap/resolution: 0.05
 * /move_base/local_costmap/robot_base_frame: base_link
 * /move_base/local_costmap/rolling_window: True
 * /move_base/local_costmap/static_map: False
 * /move_base/local_costmap/transform_tolerance: 0.5
 * /move_base/local_costmap/update_frequency: 20.0
 * /move_base/local_costmap/width: 10.0
 * /move_base/local_costmap/z_resolution: 1
 * /move_base/local_costmap/z_voxels: 2
 * /move_base/oscillation_distance: 0.5
 * /move_base/oscillation_timeout: 0.0
 * /move_base/planner_frequency: 20.0
 * /move_base/planner_patience: 5.0
 * /move_base/recovery_behavior_enabled: True
 * /move_base/shutdown_costmaps: False
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    move_base (move_base/move_base)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
ERROR: cannot launch node of type [move_base/move_base]: move_base
ROS path [0]=/opt/ros/indigo/share/ros
ROS path [1]=/home/wael/jackal_navigation/src
ROS path [2]=/opt/ros/indigo/share
ROS path [3]=/opt/ros/indigo/stacks
No processes to monitor
shutting down processing monitor...
... shutting down processing monitor complete
`
thank you in advanced

tonybaltovski, 2018-01-05:
Seems like move_base isn't installed.  Try installing the needed dependencies using `rosdep`.

wael-abdin, 2018-01-09:
thank you for your support
it worked normally after install move_base

closed 2018-01-09 by wael-abdin