jackal/jackal #57: VLP-16 in Simulation

mastash3ff, 2018-10-06:
I am trying to get the VLP-16 to work in simulation and I am having some difficulties with Jackal navigation in RViz.  I have added Jackal customizations to the *custom.urdf* file that Jackal refers to for additional accessories with the following contents:

```
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  <xacro:include filename="$(find velodyne_description)/urdf/VLP-16.urdf.xacro" />

  <bridge_plate mount="front" height="0.1"/>

  <VLP-16 parent="front_mount" name="velodyne" topic="/velodyne_points">
    <origin xyz="0 0 0.1" rpy="0 0 0" />
  </VLP-16>
</robot>
```

This results in the Jackal looking like this in Gazebo:

[image]

I would eventually like the Jackal to work off of 3D pointcloud data but I have opted for 2D Laserscan as I think it is the easier choice to get Jackal running.  With that being said, here is the launch file that does that conversion:

```
<launch>

  <param name="/use_sim_time" value="true" />

  <node pkg="tf" type="static_transform_publisher" name="velodyne_to_front_laser" args="0 0 0 0 0 0 /velodyne /front_laser 100" />

  <node pkg="pointcloud_to_laserscan" type="pointcloud_to_laserscan_node" name="pointcloud_to_laserscan">

    <remap from="cloud_in" to="/velodyne_points"/>
    <remap from="scan" to="/front/scan"/>
    <rosparam>
        transform_tolerance: 0.01
        min_height: 0.25
        max_height: 0.75

        angle_min: -3.1415
        angle_max: 3.1415
        angle_increment: 0.01
        scan_time: 0.1
        range_min: 0.9
        range_max: 130
        use_inf: true
        concurrency_level: 0
    </rosparam>

  </node>
</launch>
```

When I launch the Jackal gazebo, odom_navigation_demo, point_to_laserscan, and rviz launch files, the Jackal in RViz appears to stutter as the time freezes and picks up again indicated in the bottom-left corner of rviz.  Sample clip can be found here demonstrating this effect:  https://imgur.com/a/CeAMpTh 

Any advice on fixing this stuttering issue?

civerachb-cpr, 2020-09-08:
At first glance it looks like this might just be that the simulation is over-taxing your system.  What are the specs of the machine being used to run the simulation?

civerachb-cpr, 2020-09-08:
Just realized how old this ticket actually is.  Closing due to inactivity.  If this is still an issue, please feel free to open a new ticket.

closed 2020-09-08 by civerachb-cpr