jackal/jackal #61: Odometry shows false data

melvinpmanuel, 2019-07-23:
Hello .
This is Melvin again from University of Detroit mercy. Thanks for your advice with the battery management. I have figured out the charging procedures. After switching OFF the jackal, the charger connector needs to be connected to the charging port on the Jackal. I have to wait from 1 to 3 minutes to start the charging. Anyways it worked well for me. But if I touch or give a hit to the Jackal, the robot will automatically switch ON. (I dont know why its happening without pressing the power switch). On the charging port of robot(the female port) , I can see 4 wires going to the board. But the charger you provided have only two wires coming out. What are those two extra wires going to the board ? Is that the ground? Is that creating the above hit/touch power ON issue?

**ISSUE 2 :** 
We have purchased 5 jackals from clear path robotics and all 5 are behaving in 5 different ways.
Let me explain some of the issues we are still facing.
When we launch the "roslaunch jackal_base base.launch" the robot gets successfully up with out any issues. But when it comes to odom, it creates many problems. We are using the odom topic named " Odometry/filtered" and its shows the Position and orientation of the robot. 

For moving the robot,  we are using the " rosrun teleop_twist_keyboard teleop_twist_keyboard.py" to teleop the robot.
When we are driving the robot to X direction ( By hitting the "i" key on the keyboard"), the X position on the Odometry/filtered topic under Position. X  does not change. But at the same time the Position.Y is changing in negative direction accordingly. 
Also, when we are trying to move the robot to Y direction, the Postion.X is changing.
A screen shot of the topic is attached here. We have driven the robot to 5 meters towards X direction and took the screenshot.( Please note the "-" in the Y values)
[image]
With this issue we are not able to run any algorithm on the Jackals. This issue is there on all the five Jackals we purchased from clearpath.
Please help us to resolve this issues.
Also if possible, please send us a document which explains the connection, package installation and configuration, if we are using an external computer (in our case, Intel NUC computer instead of internal computer). We have correct documentation for simulation. But with real robots, the document available is not sufficient enough. One of the mistakes we found in your document is, the IMU tool package should be there to make the jackal_base.launch to work. But you haven’t mentioned about it anywhere in the document. 

Thanks
Melvin P Manuel
Dept. of ECE
University of Deroit Mercy

tonybaltovski, 2019-07-23:
Hi Melvin,

This issue would be better dealt with through Clearpath's support channel since it doesn't deal with the software in the repository.  I have created a support ticket for you and our support team will follow-up with you as soon as possible.

Cheers,
Tony

tonybaltovski, 2019-07-23:
However, I realized that we do not have an email to contact you at @melvinpmanuel.  If you could submit a ticket [here](https://support.clearpathrobotics.com/hc/en-us/requests/new?ticket_form_id=59673) or simply email support@clearpathrobotics.com so we can help out.

melvinpmanuel, 2019-07-24:
Thanks Tony..  I will send a mail to your support team. My email ID is manuelmp@udmercy.edu.

Regards

Melvin

closed 2019-12-11 by tonybaltovski