jackal/jackal #66: Multirobot functionality in gazebo

heuristicus, 2020-04-28:
I'm working on setting up multiple jackals to run in gazebo simulation and was wondering if there is some way to do this easily on the current branch that I've missed. I initially tried launching the jackal components in a namespace of the name of the robot (i.e. using `<group ns=$(arg robot_name)>` in a custom launch file), but it didn't work as I had hoped.

I noticed that there is the [multirobot](https://github.com/jackal/jackal/tree/multirobot) branch, which is very old, but I based some of my changes on that.

There are also two issues here (#23, #40) which mention multirobot but there hasn't been much discussion. I had a brief look at some of the other forks and a couple of them seemed to have changes to implement multirobot functionality. I think it would be a good idea to have that capability in the official repository, rather than having to maintain custom forks.

If it could be added here, I would be happy to discuss and make a PR.

civerachb-cpr, 2020-06-02:
We're working on adding multi-robot support for our simulations.  I can't give a firm estimate for when those efforts will be merged into the Jackal simulation, but it is a common enough feature request that I hope it will be ready soon.

l-schilling, 2020-12-09:
Till there is an official version released you can look at something like this: http://wiki.ros.org/multi_jackal_tutorials

closed 2022-01-11 by tonybaltovski