jackal/jackal #70: teleop_ps4.yaml overrides the dev option for joy_node

kaineub, 2020-07-21:
The teleop_ps4.yaml file that is loaded by teleop.launch overrides the dev option within the joy_node. The launch file gives the impression that passing in a joy_dev argument will let us override the default option; however, whatever option we pass in is overwritten by the yaml file (because of the last line in the yaml file). Our PS4 controller on our simulation computer uses a different port than in the yaml file and will not run with the included launch file. We had to create our own launch and yaml files to fix this issue.

Command: `roslaunch jackal_control jackal_teleop.launch dev_joy:='/dev/input/js0'`
Expected behaviour: The joy_node will use the PS4 controller located at '/dev/input/js0'
Experienced behaviour: The joy_node tries to find a PS4 controller at '/dev/input/ds4x' on the Kinetic branch or '/dev/input/ps4' on the Melodic branch

civerachb-cpr, 2020-09-08:
Good catch.  Thanks for bringing this to our attention.

civerachb-cpr, 2020-09-09:
Resolved for Melodic in https://github.com/jackal/jackal/pull/73

Will apply a similar change to Kinetic.

closed 2020-09-09 by civerachb-cpr