jackal/jackal #72: Cannot localize/navigate correctly

Tobias-Fischer, 2020-07-28:
Hi there,
We have big issues getting our robot to localize/navigate correctly, both using gmapping/amcl or with ark bridge (qut10 robot). It is unfortunate that we can not use the robot localization/navigation capabilities out-of-the-box.

There are multiple problems we noticed:
1) We have a um7_imu as well as the standard imu. For some reason the um7 launch file was configured such that the remap happens from `/imu/data_raw` to `/imu_um7/data`, which then led the `/imu` topic to contain mixed information from the internal as well as um7 imu. I fixed it by remapping `/imu` to `/imu_um7` (the whole topic), so that the external imu publishes on a separate topic
2) The internal imu does not publish meaningful data on `/imu/mag`. The values are just static.
3) The internal imu's tf frame co-incides with base_link which is very unlikely as the base_link is located at the bottom frame of the robot.
4) When using amcl, the robot seems to "overshoot", i.e. when moving forward by e.g. 1m in the physical world the robot moves 1.1m when displayed in rviz.
5) We needed to change the imu_config in `robot_localization.yaml` as otherwise there was a massive movement in z-direction, from
```
imu0_config: [false, false, false,
              true, true, true,
              false, false, false,
              true, true, true,
              false, false, false]
```

to
```
imu0_config: [false, false, false,
              false, false, true,
              false, false, false,
              false, false, true,
              false, false, false]
```

6) When trying to use the um7 imu in robot_localization, massive errors occur. I think that the um7_imu tf link might be wrongly oriented.
7) When doing the imu compass calibration, the calibration node writes a calibration file where `use_mag` is true, which leads to even bigger errors due to 2)

It would be great if you could let me know how to fix our issues.

Many thanks,
Tobias

prefpkg21, 2020-09-04:
They disable the internal compass. It received too much interference from the motors. You could use a Bosch BNo055 imu that has a compass, what we did

civerachb-cpr, 2020-09-08:
Sorry to hear you're having problems with your robot's initial configuration.

As mentioned above, typically we disable the magnetometer, as the interference from the chassis and motors can make this data very unreliable.

It's possible that whoever did the integration on your robot misconfigured it and the issues weren't caught before the robot was shipped out.  If this is the case, we sincerely apologize!

Have you contacted our customer & technical support team?  They'll be in a better position to diagnose & resolve the problems with your specific robot, rather than going through GitHub.  You can find the contact information for our technical support channel here: https://clearpathrobotics.com/talk-to-a-human/

civerachb-cpr, 2021-01-28:
Closing due to inactivity.  If this issue is ongoing please feel free to comment and we can re-open it.

closed 2021-01-28 by civerachb-cpr