jackal/jackal #76: Issues with the networking

utsavpatel22, 2020-12-14:
Hi, 
We bought Clearpath Jackal a year back. We had reinstalled the OS on the system due to some issues we faced with the ROS packages. After reinstalling the OS, we are facing issues with the networking. The Ethernet IP is not stable, so many times the default Ethernet IP (192.168.131.1) doesn't work. I tried to directly connect to the non static Ethernet IP but that always doesn't work. I am copying the /etc/network/interfaces file here, 

auto lo br0 br0:0
iface lo inet loopback

# Bridge together physical ports on machine, assign standard Clearpath Robot IP.
iface br0 inet static
  bridge_ports regex (eth.*)|(en.*)
  address 192.168.131.1
  netmask 255.255.255.0
  bridge_maxwait 0

auto wlan0
iface wlan0 inet dhcp
                wpa-ssid NETGEAR21-5G
                wpa-psk **************

# Also seek out DHCP IP on those ports, for the sake of easily getting online,
# maintenance, ethernet radio support, etc.
iface br0:0 inet dhcp

civerachb-cpr, 2020-12-14:
Can you run `wicd-curses` and open the preferences by pressing shift + P.  The "Wired Interface" section should be blank.  If there's something in it (e.g. "br0" or "en0s2") delete it, save your preferences, and reboot the robot.  Sometimes wicd tries managing the wired interface, even though it's configured through network/interfaces.

Similarly, if you're manually configuring the wireless network through the interfaces file, you may want to disable managing the wireless through wicd too.

utsavpatel22, 2020-12-14:
Thanks @civerachb-cpr for responding. I removed "en0s2" from the "Wired Interface" and rebooted the machine, still I cant connect to the static IP 192.168.131.1.

civerachb-cpr, 2020-12-14:
When you say "cannot connect" how are you trying to connect?  Is the robot connected to a router or a switch? Or do you have an ethernet cable connected directly from your computer to the second ethernet port on the robot?

utsavpatel22, 2020-12-14:
I have the Ethernet cable directly from the robot (Static Port) to the PC.

civerachb-cpr, 2020-12-14:
Because of the bridged network it doesn't matter which port you connect to: either one can be used to access the static network inside the robot.

How have you configured your PC's ethernet port?  Because the robot does not have a DHCP server running on it you cannot just plug the cable into your PC; you need to configure your PC's wired network to use a static IP address with the following parameters:

```
address: 192.168.131.200   # technically anything on the 192.168.131.* subnet will work, but some addresses <=100 are reserved for sensors, MCU hardware, etc...
netmask: 255.255.255.0
```

You can omit setting the gateway, as this isn't configuring an internet-facing network connection.

utsavpatel22, 2020-12-14:
Thanks, @civerachb-cpr for the detailed answer. On my laptop, I have set up the static IP according to the instructions given in the Jackal's manual. The static IP is set to 192.168.131.51. The main issue is that the connection is very unstable, sometimes I can ssh to Jackal using the IP 192.168.131.1 and other times I can't connect. If I run "ifconfig" on Jackal (By connecting a screen directly to Jackal) I don't see "br0" listed.

civerachb-cpr, 2020-12-14:
Did you install the Jackal software using the Clearpath ISOs (https://packages.clearpathrobotics.com/stable/images/latest/), or did you use a standard Ubuntu image?

utsavpatel22, 2020-12-15:
I am using the Clearpath ISO.

civerachb-cpr, 2020-12-15:
Can you please confirm that the bridge-utils package is installed on the robot?

utsavpatel22, 2020-12-15:
Yes, it is installed.

utsavpatel22, 2020-12-15:
I was going through some articles, br0 bridge was disabled. I enabled it, it is bridging the right ports but "ifconfig" doesnt show the static IP 192.168.131.1 for br0.

civerachb-cpr, 2020-12-15:
Can you run `sudo ifup br0`?  It may be that the interface just isn't coming up for whatever reason.  Worst-case you can work-around the problem by forcing the interface to come up by putting `ifup br0` in `/etc/rc.local`

utsavpatel22, 2020-12-15:
I ran `sudo ifconfig br0 up` and it worked. But after it is up, I can see it in the list provided by `ifconfig` but it doesnt have an IP address, it is in UP state though. Whenever the system ran I could see the IP 192.168.131.1 under `br0`. I have attached a screenshot. 
[image]

civerachb-cpr, 2020-12-15:
Is this screenshot taken from directly on the robot, or are you SSH'd in from an Ubuntu desktop installation?

utsavpatel22, 2020-12-15:
This screenshot is taken from my laptop that is connected to the robot using a direct Ethernet connection. I am using `enp3s0` interface that has the IP `169.254.81.20`

civerachb-cpr, 2020-12-16:
If your laptop is connected to the robot directly your laptop should have a statically-configured IP address on the 192.168.131.* address space.  If your robot and laptop are both connected to a router then you can use whatever DHCP address the router assigned, and log into the robot using the robot's assigned DHCP address.

utsavpatel22, 2020-12-17:
I have configured a static IP on my laptop, but when I connect the laptop to the robot, it doesn't show the robot's IP which is supposed to be `192.168.131.1` and I SSH into the robot using the IP.

civerachb-cpr, 2020-12-17:
When you say "it doesn't show the robot's IP" what is "it" in this context?

utsavpatel22, 2020-12-17:
When I connect the robot to the laptop after setting up the static IP on the laptop, the laptop doesn't show the IP of the robot that is supposed to be `192.168.131.1`. And if I try to SSH from my laptop to the robot (on IP `192.168.131.1`) the laptop doesnt connect to the robot. On the robot, if I run `ifconfig`, that also doesnt show the bridge IP `192.168.131.1` under `br0`.

civerachb-cpr, 2020-12-17:
On the robot, can you please run 

```
sudo ifdown br0
sudo ifup br0
```

and then run

```
ifconfig br0
```

utsavpatel22, 2020-12-17:
@civerachb-cpr thank you, it worked. Should I just put `ifup br0` in `/etc/rc.local`?

civerachb-cpr, 2020-12-17:
Yes, you can do that.  In an ideal world you wouldn't need to do that, but forcing the network to come up with rc.local is fine if it gets around the problem and lets you get back to working with your robot.

utsavpatel22, 2020-12-17:
Thank you, @civerachb-cpr It works. Closing the issue.

closed 2020-12-17 by utsavpatel22