jackal/jackal #77: the error when roslaunch jackal_gazebo jackal_world.launch config:=front_laser

gaopenggang, 2020-12-23:
Hi,
when  I run your code of kinetic-devel in the linux 16.0 system and the gazebo vision is 7, there are some problem.
after roslaunch jackal_gazebo jackal_world.launch config:=front_laser, the error is blow:
```
process[gazebo-1]: started with pid [619]
process[gazebo_gui-2]: started with pid [624]
process[robot_state_publisher-3]: started with pid [629]
process[controller_spawner-4]: started with pid [630]
process[ekf_localization-5]: started with pid [631]
process[twist_mux-6]: started with pid [642]
process[bluetooth_teleop/joy_node-7]: started with pid [666]
process[bluetooth_teleop/teleop_twist_joy-8]: started with pid [686]
process[twist_marker_server-9]: started with pid [693]
process[urdf_spawner-10]: started with pid [700]
[ERROR] [1608713418.941996037]: Couldn't open joystick /dev/input/ds4x. Will retry every second.
[ INFO] [1608713419.235253562]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1608713419.236502958]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1608713419.395794919]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1608713419.396990136]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
Warning [parser_urdf.cc:1232] multiple inconsistent <gravity> exists due to fixed joint reduction overwriting previous value [true] with [false].
Warning [parser_urdf.cc:1232] multiple inconsistent <gravity> exists due to fixed joint reduction overwriting previous value [false] with [true].
[urdf_spawner-10] process has finished cleanly
log file: /home/gao/.ros/log/9a46f524-44c4-11eb-9e40-e4f89ca14e1a/urdf_spawner-10*.log
[ INFO] [1608713423.118723194, 1517.347000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1608713423.118860740, 1517.347000000]: Starting Laser Plugin (ns = /)
[ INFO] [1608713423.120948230, 1517.347000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1608713423.247366724, 1517.347000000]: Loading gazebo_ros_control plugin
[ INFO] [1608713423.247517802, 1517.347000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1608713423.249640013, 1517.347000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1608713423.579833183, 1517.347000000]: Loaded gazebo_ros_control.
[ INFO] [1608713423.757821729, 1517.367000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1608713423.758560951, 1517.367000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: receive_once[/controller_manager/load_controller]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
[WARN] [1608713423.831634, 1517.401000]: Controller Spawner error while taking down controllers: transport error completing service call: receive_once[/controller_manager/switch_controller]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'
[ INFO] [1608713423.976212307, 1517.460000000]: Physics dynamic reconfigure ready.
[ INFO] [1608713424.009957234, 1517.476000000]: Physics dynamic reconfigure ready.
[controller_spawner-4] process has died [pid 630, exit code 1, cmd /opt/ros/kinetic/lib/controller_manager/spawner jackal_joint_publisher jackal_velocity_controller __name:=controller_spawner __log:=/home/gao/.ros/log/9a46f524-44c4-11eb-9e40-e4f89ca14e1a/controller_spawner-4.log].
log file: /home/gao/.ros/log/9a46f524-44c4-11eb-9e40-e4f89ca14e1a/controller_spawner-4*.log

```
Most importantly ,I **can not find the wheel of robot model in the rviz**,  after run navigation and rviz package.
tf errer is **No transform from [front_left_wheel_link] to [odom]**.
Do you have any suggestion about it? Thanks

civerachb-cpr, 2020-12-23:
Does this error also appear if you omit the "config:=front_laser" argument from the command-line?

gaopenggang, 2020-12-23:
Yes, it also appears.

gaopenggang, 2020-12-23:
> Does this error also appear if you omit the "config:=front_laser" argument from the command-line?

yes,it also appears.

civerachb-cpr, 2020-12-23:
It sounds like you're either missing a package or an update for a package.

Can you please check for updates and install them by running `sudo apt-get update` followed by `sudo apt-get upgrade` (and possibly `sudo apt-get dist-upgrade` if upgrade indicates you have held packages)

If you built jackal_gazebo from source, can you please make sure you have the `kinetic-devel` branches checked-out (instead of `melodic-devel`), `cd` into your catkin workspace and re-run `rosdep install --from-paths src --ignore-src -r -y`?

gaopenggang, 2020-12-24:
> It sounds like you're either missing a package or an update for a package.
>
> Can you please check for updates and install them by running `sudo apt-get update` followed by `sudo apt-get upgrade` (and [...]

Thanks firstly, but it does not solve my problem. I built jackal_gazebo from source of **kinetic-devel** branches and run `rosdep install --from-paths src --ignore-src -r -y` then in the catkin workspace. 
the result is 
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
jackal_robot: Cannot locate rosdep definition for [jackal_firmware]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```
I guess this error is not relevant to my error of jackal_gazebo or jackal_description, so ignore it.
There are 12 packages in my workspace, shown below:
```
-- ~~  - jackal_bringup
-- ~~  - jackal_desktop (metapackage)
-- ~~  - jackal_firmware_ms
-- ~~  - jackal_robot (metapackage)
-- ~~  - jackal_simulator (metapackage)
-- ~~  - jackal_msgs
-- ~~  - jackal_control
-- ~~  - jackal_description
-- ~~  - jackal_gazebo
-- ~~  - jackal_navigation
-- ~~  - jackal_viz
-- ~~  - jackal_base
```
Now, do you have some suggestion about it?

civerachb-cpr, 2020-12-24:
For the jackal_firmware issue, you can solve that by following the instructions on packages.clearpathrobotics.com and adding the apt and rosdep sources as described there.  You are correct though, jackal_firmware is not needed for the jackal_gazebo nor jackal_description packages; it's a dependency of jackal_robot, and is only needed when working with a physical robot.

Did you check if there were any updated apt packages available on your system?  Last time we had reports about this sort of problem it was caused by a third-party dependency (genpy in that instance), which subsequently released an update that needed to be installed.

gaopenggang, 2020-12-28:
Thanks a lot. it worked after run `apt-get update` and `apt-get upgrade`.

closed 2020-12-28 by gaopenggang