jackal/jackal #129: Gazebo Laser Plugin

Cavalletta98, 2023-02-03:
Hi, i'm running the simulation on Ubuntu 20.04 with ROS Foxy. I noticed that it is not possible to simulate a Laserscan because it is not present as gazebo plugin with name `libgazebo_ros_laser.so`. In ROS2, should be `libgazebo_ros_ray_sensor.so`.
Can you please help me to simulate it?
Thanks

tonybaltovski, 2023-02-03:
@Cavalletta98 can you try `foxy-devel` from source as #130 should fix your issue.

Cavalletta98, 2023-02-03:
Is It updated also the apt package?

tonybaltovski, 2023-02-08:
I have a made a new release of `1.0.3` with the fix.  It will be on Open Robotics' package server at the next sync and on [Clearpath's package](http://wiki.ros.org/ClearpathRobotics/Packages) server soon.

PR #130 (2023-02-03): Updated gazebo_ros_laser plugin
Fixes #129.
closed 2023-02-08 by tonybaltovski