jackal/jackal #132: Problem displaying laser scan

Cavalletta98, 2023-02-14:
Hi, i'm running on Ubuntu 20.04 with ROS Foxy and i'm trying to echo the laser topic with the command `ros2 topic echo /front/scan`. I'm getting this error `Cannot echo topic '/front/scan', as it contains more than one type: [sensor_msgs/msg/LaserScan, sensor_msgs/msg/PointCloud2].`. Can you please help me?
Thanks

Cavalletta98, 2023-02-14:
i run the command `export JACKAL_LASER=1`

Cavalletta98, 2023-02-14:
I have updated the comment. Sorry for not beeing precise

tonybaltovski, 2023-02-14:
Thanks! I was able to reproduce the issue and the fix is #133. Please give that a try.

Cavalletta98, 2023-02-14:
The fix worked for the echo command. When i run my launch file provided to you before, i got this error `[controller_server-2] [INFO] [1676395444.311756939] [local_costmap.local_costmap_rclcpp_node]: Message Filter dropping message: frame 'front_laser' at time 1573.970 for reason 'Unknown'`

Cavalletta98, 2023-02-15:
@tonybaltovski I noticed that if i launch the simulation with `ros2 launch jackal_gazebo jackal_world.launch.py` and i do `ros2 topic echo /front/scan `, i'm able to see the laser scan values. But, if i also run my launch file, i cannot see the laser scan values with the command `ros2 topic echo /front/scan `. Probably there is something wrong in the [navigation config](https://github.com/jackal/jackal/blob/foxy-devel/jackal_navigation/config/nav2.yaml)

tonybaltovski, 2023-02-15:
Can you set `use_sim_time` to `true`? Can you provide the output of when you launch your launch file as well as `ros2 launch jackal_gazebo jackal_world.launch.py`?

Cavalletta98, 2023-02-16:
I setted `use_sim_time` to `true`. I attached the two logs files
[my_launch_file_log.txt](https://github.com/jackal/jackal/files/10753645/my_launch_file_log.txt)
[Jackal_world_launch_log.txt](https://github.com/jackal/jackal/files/10753646/Jackal_world_launch_log.txt)

tonybaltovski, 2023-02-16:
We are able to reproduce the issue and will get back to you.

Cavalletta98, 2023-02-16:
Thanks. I think the two issue are related. I was able to visualize the robot in the map
frame in rviz but not able to show the map built from the slam toolbox

DrAndyWest, 2023-02-23:
Hi All,
I have had this issue with the Jackal in sim on Foxy.  I have found that the timestamps don't seem to match between the robot state publisher and the laser scan.  The laser scan appears to be in sim time (small numbers when ros2 topic echo), whereas other parts are using wall time (clearly epoch values, e.g. using tf viewer in rqt).  Running RVIZ gladly shows the scan in base_link frame, but slam_toolbox can't produce maps as a result.

Cavalletta98, 2023-02-25:
@DrAndyWest HAve you solved it?

DrAndyWest, 2023-03-06:
Hi All,
I have managed to get things working.  I found a few launch files/nodes not using or passing use_sim_time=true, this leads to the robot sim TF and nodes having different timestamps to the gazebo sensors.  Now SLAM Toolbox can generate a map from the lidar scan.

**jackal_gazebo/gazebo.launch.py**
[Line 95](https://github.com/jackal/jackal_simulator/blob/49039fce6091fabc6535aa18942713d1a4266c6a/jackal_gazebo/launch/gazebo.launch.py#L95) and [Line 59](https://github.com/jackal/jackal_simulator/blob/49039fce6091fabc6535aa18942713d1a4266c6a/jackal_gazebo/launch/gazebo.launch.py#L59) call launch files, parameters can be successfully passed using a dictionary rather than list `launch_arguments={'robot_description_command': robot_description_command,'is_sim':'True',}.items()`.  No need to add additional parameters to 'spawn_jackal' for example.

**jackal_description/description.launch.py**
Does not have an `is_sim` argument, e.g.
```
is_sim_arg = DeclareLaunchArgument('is_sim', default_value='False')
is_sim = LaunchConfiguration('is_sim')
# Declare the launch argument using
ld.add_action(is_sim_arg)
# or
# ld = LaunchDescription([is_sim_arg])
```
The [robot_state_publisher node line](https://github.com/jackal/jackal/blob/e1342033f83d08f1455463ce2533cad06d1d5b07/jackal_description/launch/description.launch.py#L28) needs to include use_sim_time, e.g.`parameters=[{'use_sim_time': is_sim}, {'robot_description': robot_description_content},]`

**jackal_control/teleop_base.launch.py**
Needs to have an is_sim argument added (same as above)
Arguments need to be passed to the [interactive_marker_twist_server node](https://github.com/jackal/jackal/blob/e1342033f83d08f1455463ce2533cad06d1d5b07/jackal_control/launch/teleop_base.launch.py#L21) and the [twist_mux node](https://github.com/jackal/jackal/blob/e1342033f83d08f1455463ce2533cad06d1d5b07/jackal_control/launch/teleop_base.launch.py#L30).  However, this might be unnecessary.

**jackal_control/control.launch.py**
This was the major thing to fix.
Add use_sim_time parameter to [ekf_node](https://github.com/jackal/jackal/blob/e1342033f83d08f1455463ce2533cad06d1d5b07/jackal_control/launch/control.launch.py#L63) and [imu_filter node](https://github.com/jackal/jackal/blob/e1342033f83d08f1455463ce2533cad06d1d5b07/jackal_control/launch/control.launch.py#L72).

Annoyingly, the control nodes cannot have use_sim_time arguments passed to them using launch arguments ([see this issue](https://github.com/ros-controls/ros2_control/issues/325#issuecomment-879355414)).  We must amend the jackal_control/config/control.yaml file.

**jackal_control/config/control.yaml**
```
controller_manager:
  ros__parameters:
    use_sim_time: true
    update_rate: 50  # Hz
```
User beware!  I noticed that the Jackal hardware uses this control code too ([see here](https://github.com/jackal/jackal_robot/blob/7619eb3718fe232447bb746e014bd6d1662796a9/launch/bringup.launch.py#L39)).  Therefore, hard-coding in use_sim_time: true is kind of risky.  As I use both sim and hardware I will have to be careful.

As this is a decision to be made by Clearpath how to handle this use_sim_time issue for the controller manager (which gets passed to jackal_velocity_controller and joint_state_broadcaster), I don't think a PR request and pushing my changes is sensible.