jackal/jackal #138: [Question] Radius for Jackal's Nav2 Global and Local Costmap footprints?

brendoncintas, 2023-05-23:
**Please provide the following information:**
 - OS: Ubuntu 20.04
 - ROS Distro: Foxy
 - Built from source or installed: Built from source
 - Package version: 79fc00f49a4309dffe0cc9d6e9621f081a0e0134
 - Real hardware or simulation: Yes

 **Expected behaviour**
Local and global costmap footprints for the robot should be rectangular.

 **Actual behaviour**
Local and global costmap footprints for the robot are actually circular in lines [169 local costmap](https://github.com/jackal/jackal/blob/45daab215c95c262afce11a41b9a77ab2092d56f/jackal_navigation/config/nav2.yaml#L169) and [209 - global costmap](https://github.com/jackal/jackal/blob/45daab215c95c262afce11a41b9a77ab2092d56f/jackal_navigation/config/nav2.yaml#L209), which I believe is affecting the robot's ability to perceive its ability to navigate. 

**To Reproduce**
Provide the steps to reproduce:
1. Run Nav2 using `ros2 launch nav2_bringup navigation_launch_py use_sim_time:=true params_file:=<path/to/jackal_navigation/>nav2.yaml
2. Open Rviz and add local costmap polygon feature in Rviz2

**Other notes**
Love the work you guys are doing; I'm having a blast futzing around with this Jackal simulator.

tonybaltovski, 2023-05-29:
@brendoncintas good catch.  We'll change to use a rectangular footprint.

brendoncintas, 2023-05-29:
@tonybaltovski I reviewed your PR for #139. I did something similar, but my dimensions were slightly different (see [here](https://github.com/jackal/jackal/blob/81fe7931e9e7670a0084afa526243ce7644f1fd9/jackal_navigation/config/nav2.yaml#L169)). 

I took the measurements from the Jackal documentation shown [here](https://clearpathrobotics.com/jackal-small-unmanned-ground-vehicle/), using wheel-to-wheel (sides) and bumper-to-bumper (forward and back), then divided them by 2 in order to take create the points relative to the centroid of the robot.

tonybaltovski, 2023-05-30:
@brendoncintas you are correct.  I re-used the footprint from [Noetic](https://github.com/jackal/jackal/blame/noetic-devel/jackal_navigation/params/costmap_common_params.yaml#L11).  Want to do a PR for the correct footprint?

brendoncintas, 2023-05-30:
@tonybaltovski 

Will do. Sending a pull request shortly.