jackal/jackal #144: Need Help with Jackal Navigation Issues: Misalignment and Erratic Movement

MohamedHabbad, 2023-07-04:
Hey everyone,
i'm currently facing some challenges with the navigation setup of my Jackal robot and i need some guidance from you. Here's a breakdown of the issues that i am experiencing and the steps i've been taken so far:

**Mapping and Localization:**
   - I followed the instructions provided in https://docs.clearpathrobotics.com/docs/robots/outdoor_robots/jackal/tutorials_jackal/tutorials_jackal_ros1/ to create a map using gmapping with my Jackal robot.
   - To navigate with the map, I launched the following two files:
     - On the Jackal: `roslaunch jackal_navigation amcl_demo.launch map_file:=/home/administrator/mymap.yaml`
     - On my computer: `roslaunch jackal_viz view_robot.launch config:=localization`
   - Initially, the laser scanner's placement on the map did not match its physical position, which i already expected, see this picture :
[image]

   - After giving an approximate location for the robot, the visualization improved and looked better see those pic:
   
[image]
[image]
[image]

**Misalignment Issue:**
   - However, when I send the robot to different locations, there are instances where the localization becomes misaligned with the map, as shown in this photo:
   
[image]

or this :

[image]

   - I'm unsure why this misalignment occurs, especially after the initial successful localization !!?
   
**Erratic Movement:**

   - Additionally, during navigation, i've observed that the robot moves in a backward and forward pattern multiple times before resuming normal movement!
   - This erratic behavior can be seen in this video that i took :

https://github.com/jackal/jackal/assets/136476747/a8dc6b6a-792a-4e6e-93f7-9d8a6d2f9985

I would greatly appreciate any explanations, or troubleshooting suggestions regarding these issues. 
Thank you in advance for your help

civerachb-cpr, 2023-07-05:
Does your robot have an external IMU (e.g. a GX5 or UM7 IMU) or are you only using the IMU built into the Jackal's MCU?  Using an external IMU will greatly improve the odometry data, which may help address the drifting between the lidar and map data.

It's not uncommon for the lidar information to drift slightly compared with the map data while the robot is in motion, especially if you're running Rviz on a remote computer with a wireless connection.  When the robot comes to a stop do the lidar lines re-align themselves with the map?

Regarding the erratic movement, that's not uncommon when using AMCL and Gmapping; these are relatively simple navigation packages, and they will sometimes move erratically, especially when moving close to obstacles.  This isn't anything specific to the Jackal and is related to the underlying third-party navigation packages.

MohamedHabbad, 2023-07-06:
Thank you for your answer,no i don't have an external IMU for my robot i am currently using the IMU that is built into the Jackal itself.

Regarding your question about the lidar lines realigning with the map at the end of the Jackal's movement, unfortunately, that is not the case. In fact, I have noticed that the situation worsens when the Jackal comes to a stop. i have documented this issue with a series of photos taken over a 10-minute period :
[image]

This photo demonstrates the progression of the problem. As you can see, after I stopped the Jackal, the alignment between the lidar lines and the map deteriorated rapidly. Within just 2-3 minutes, there was a noticeable degradation, and it continued to worsen as time passed.

To provide you with more visual evidence, I have recorded this video that demonstrates the situation.

https://github.com/jackal/jackal/assets/136476747/06f9b0d9-7c6d-49ba-8bf6-7b6dd755ded3

In the video, you will notice that initially, when I drive the robot slightly forward, the map and scanner lines match perfectly and show improvement compared to the initial estimated pose. However, as I drive the robot to the other side of the room, the alignment worsens.

Thank you once again.

civerachb-cpr, 2023-07-06:
It looks like your Jackal may be having some odometry issues.  This could be the result of a hardware problem, or it may be a configuration problem with your Jackal specifically.

Since Github issues are normally for addressing software bugs in the code in the repo, I'm going to close this issue and suggest that you [contact Clearpath's customer support](https://support.clearpathrobotics.com/hc/en-us/requests/new). Please include a link to this github issue so the support team can see the photos & videos.

closed 2023-07-06 by civerachb-cpr