jackal/jackal #146: Cannot observe any topics being published when Jackal is powered on from the PC.

ans2568, 2023-09-09:
**Please provide the following information:**
 - OS: ubuntu 20.04
 - ROS Distro: Foxy
 - Built from source or installed: Built from source
 - Package version: 
 - Real hardware or simulation: Real hardware

 **Expected behaviour**
Jackal and the PC are connected via WiFi, and I want to access Jackal's ROS2 data from the PC.

 **Actual behaviour**
Jackal and the PC are connected via WiFi, and both have ROS_DOMAIN_ID = 0. 
However, I am unable to access Jackal's ROS2 data (e.g., /cmd_vel) from the PC.

Furthermore, when I attach a LiDAR sensor to Jackal and attempt to use point cloud data within Jackal, another node inside Jackal needs to subscribe to the /pointcloud topic. However, I cannot locate this topic.

**To Reproduce**
Provide the steps to reproduce:
1. Turn on Jackal
2. Running the ping command from the PC to Jackal's computer  **Success**
3. Running the ping command from the Jackal's computer to PC  **Success**
4. Execute the following command on the PC. ```ros2 topic list```
5. I can't see topics other than the default ones like /parameter_events and /rosout.

**Other notes**
After upgrading Jackal to Ubuntu 20.04, installing ROS Foxy, and performing the firmware update.
referenced link: https://clearpathrobotics.com/assets/guides/foxy/jackal/JackalInstallRobotSoftware.html

I am unable to see the topics from a PC on the same network.

When I manually created and executed a node within Jackal to publish on the std_msgs/String topic, I could observe it from the PC. However, I cannot see the topics that are published when the Jackal is powered on and the bringup process is executed.

Additionally, I attempted to control Jackal by publishing to the /cmd_vel topic from the PC. While I confirmed that /cmd_vel is being received by Jackal, Jackal did not move.

tonybaltovski, 2023-09-11:
@ans2568 can you see the Jackal topics on the Jackal computer?

ans2568, 2023-09-11:
> @ans2568 can you see the Jackal topics on the Jackal computer?

Yes, I can see the Jackal topics on the Jackal computer

tonybaltovski, 2023-09-11:
@ans2568 Are you using FastDDS  check with `ros2 doctor --report | grep middleware`?

ans2568, 2023-09-11:
> @ans2568 Are you using FastDDS check with `ros2 doctor --report | grep middleware`?

I haven't tried this before, but I just attempted it, and here's the output.

```
middleware name     : rmw_fastrtps_cpp
```

tonybaltovski, 2023-09-13:
Is that from the Jackal or your laptop? Ensure both match.

ans2568, 2023-09-14:
> Is that from the Jackal or your laptop? Ensure both match.

Both were rmw_fastrtps_cpp, but it didn't work.

I have resolved the issue.

It appears to be an issue with the bringup that starts when the Jackal boots. The exact cause could not be determined. 
However, after uninstalling the behavior described in the link(https://clearpathrobotics.com/assets/guides/foxy/jackal/JackalInstallRobotSoftware.html)
`ros2 run jackal_robot install`, 
I proceeded to run it directly with `ros2 launch jackal_robot bringup.launch.py` and I confirmed that it worked without any issues.

Thanks for comment

closed 2023-09-14 by ans2568