jackal/jackal #148: Jackal shutting down itself

tjl98, 2023-11-01:
I've been working with the jackal for a while now and have observed that it tends to power off on its own after running for about 15-20 minutes. Has anyone else encountered this issue, and if so, how did you go about resolving it? Thank you!

civerachb-cpr, 2023-11-01:
Is the robot shutting down after 15-20 minutes of active driving, or 15-20 minutes of sitting idle?

What sensors and other accessories do you have mounted on the robot? e.g. how many lidars, is there an arm mounted to it, etc...?

tjl98, 2023-11-02:
Robot shut down after 15-10 of sitting idle. No accessories or sensors are mounted on the robot. Could it be part of the default set up?

tonybaltovski, 2023-11-02:
Can you please provide more details including:

1. your installed packages using ` apt list --installed | grep jackal`
2. which compute solution is equipped
3. the output of ` journalctl -b -1`

tjl98, 2023-11-27:
Hi! Hope this helps
Compute solution:
[image]
packages installed:
[image]

tonybaltovski, 2025-01-30:
@tjl98 I believe this was taken care by email?

closed 2025-02-03 by tonybaltovski