jackal/jackal #152: Seeting a goal as gps position

celikemir, 2024-06-06:
Hi,

Is it possible to give gps coordinate goal to jackal navigation ? I also wonder that if I provide odometry for jackal navigation and will i be able to use path planning feature of jackal navigation ?

civerachb-cpr, 2024-06-07:
This isn't a feature native to the Jackal. To accomplish this you would need to use additional third-party packages to convert GPS coordinates into odometry poses. You'd likely also need to augment the Jackal's built-in GPS with a higher-precision unit or -- ideally -- two GPS units with their antennas separated by at least 1m in order to accurately determine the robot's heading when stationary.

closed 2024-06-07 by civerachb-cpr

celikemir, 2024-06-07:
thank you for your answer. I want to be sure that suppling odometry from another ros package, without using aml package can jackal navigation be work and managed to follow trajectory ?

civerachb-cpr, 2024-06-07:
If you're supplying odometry data with another package you'll likely want to disable the Jacka's default odometry. You can do this by adding

```
export ENABLE_EKF=false
```
to `/etc/ros/setup.bash` and rebooting the robot. This will disable the EKF node that runs by default, allowing you to use a different odometry package.

Whatever software you're using to command the robot should output a `geometry_msgs/Twist` messages on the `/cmd_vel` topic; this will be picked up by the Jackal's velocity controller and command the motors according to the input commands.

The Jackal has no built-in collision-avoidance nor obstacle detection; it is the responsibility of whatever is publishing velocity commands to handle this appropriately. Normally this is done by subscribing to 2D or 3D lidar data and building a local costmap, but other solutions are possible.