jackal/jackal #155: how to receive data on the /scan topic?

Ji-Sung05, 2025-01-07:
Hello, I am currently running a Jackal simulation on Foxy.
I am trying to generate a map using SLAM Toolbox. However, after adding a LiDAR (raider) sensor to the jackal.urdf.xacro file and running SLAM Toolbox, I encountered an issue where no data is being received from the /scan topic.

It seems that the LiDAR plugin might not be applied correctly. Could you please guide me on how to receive data on the /scan topic?

Below, I have attached the jackal.urdf.xacro file with the LiDAR plugin added.

```
  <!-- LIDAR Link -->
<link name="lidar_link">
  <visual>
    <origin xyz="0 0 0" rpy="0 0 0" />
    <geometry>
      <cylinder radius="0.02" length="0.05" />
    </geometry>
    <material name="black" />
  </visual>
  <collision>
    <origin xyz="0 0 0" rpy="0 0 0" />
    <geometry>
      <cylinder radius="0.02" length="0.05" />
    </geometry>
  </collision>
  <inertial>
    <origin xyz="0 0 0" rpy="0 0 0" />
    <mass value="0.1" />
    <inertia ixx="0.0001" ixy="0.0" ixz="0.0" iyy="0.0001" iyz="0.0" izz="0.0001" />
  </inertial>
</link>

<!-- LIDAR Joint -->
<joint name="lidar_joint" type="fixed">
  <origin xyz="0 0 0.2" rpy="0 0 0" /> <!-- LiDAR 위치 -->
  <parent link="chassis_link" />
  <child link="lidar_link" />
</joint>

  <!-- LIDAR Joint -->
  <gazebo reference="lidar_link">
    <sensor name="lidar_sensor" type="ray">
      <pose>0 0 0 0 0 0</pose>
      <visualize>true</visualize> <!-- 시각화를 활성화 -->
      <update_rate>20</update_rate>
      <ray>
        <scan>
          <horizontal>
            <samples>720</samples>
            <resolution>1</resolution>
            <min_angle>-1.5708</min_angle> <!-- -90도 -->
            <max_angle>1.5708</max_angle>  <!-- 90도 -->
          </horizontal>
        </scan>
        <range>
          <min>0.2</min>
          <max>10.0</max>
          <resolution>0.01</resolution>
        </range>
      </ray>
      <plugin name="gazebo_ros_ray_sensor" filename="libgazebo_ros_ray_sensor.so">
        <ros>
          <namespace>/</namespace>
          <remapping>~/out:=/scan</remapping> <!-- ROS2 토픽 이름 -->
        </ros>
        <topic_name>/scan</topic_name>
	      <frame_name>lidar_link</frame_name>
	      <output_type>sensor_msgs/LaserScan</output_type>
      </plugin>
    </sensor>
  </gazebo>
```

Ji-Sung05, 2025-01-07:
I have modified the current code and am successfully receiving data from the /scan topic. 
However, when I run `ros2 launch slam_toolbox online_async_launch.py use_sim_time:=true`, the following error log appears:
```
[async_slam_toolbox_node-1] [INFO] [1736217721.381889397] [slam_toolbox]: Message Filter dropping message: frame 'base_link' at time 2138.404 for reason 'Unknown'
```

Additionally, I am unable to receive data from the /map topic. How can I resolve this issue?

tonybaltovski, 2025-01-15:
Is this being done with an off-board computer?  If so, have you setup time syncing?  https://docs.clearpathrobotics.com/docs/ros/networking/ntp

Ji-Sung05, 2025-01-16:
I am working with Gazebo simulations instead of a real robot
Even after running the command `ros2 launch slam_toolbox online_sync_launch.py use_sim_time:=true` does not resolve the issue.

and manually modifying the relevant files to set use_sim_time=true, the timestamp synchronization still doesn't work.

The file I modified is online_async_launch.py

Ji-Sung05, 2025-01-17:
When running the Jackal Gazebo simulation, I observed that it was initialized with the current time instead of the simulation time. Upon investigation, I identified that this issue occurs because the libgazebo_ros_init.so plugin is not being loaded during the simulation. To address this, I modified the gazebo.launch.py file as shown below. However, despite these changes, the simulation still initializes with the current time rather than the simulation time.

I followed the Jackal 1.0.0 tutorial step by step. However, when I run the Jackal simulation, the simulation time is not applied. Could you  provide a solution to resolve the issue? If this doesn't work, I plan to switch to Humble.

```
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, ExecuteProcess, IncludeLaunchDescription, SetEnvironmentVariable
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import EnvironmentVariable, FindExecutable, LaunchConfiguration, PathJoinSubstitution

from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

from pathlib import Path

from ament_index_python.packages import get_package_share_directory

ARGUMENTS = [
    DeclareLaunchArgument('world_path', default_value='',
                          description='The world path, by default is empty.world'),
    DeclareLaunchArgument('use_sim_time', default_value='true', description='Use simulation time'),
]

def generate_launch_description():

    gz_resource_path = SetEnvironmentVariable(name='GAZEBO_MODEL_PATH', value=[
                                                EnvironmentVariable('GAZEBO_MODEL_PATH',
                                                                    default_value=''),
                                                '/usr/share/gazebo-11/models/:',
                                                str(Path(get_package_share_directory('jackal_description')).
                                                    parent.resolve())])

    # Launch args
    world_path = LaunchConfiguration('world_path')
    use_sim_time = LaunchConfiguration('use_sim_time')
    prefix = LaunchConfiguration('prefix')

    config_jackal_velocity_controller = PathJoinSubstitution(
        [FindPackageShare('jackal_control'), 'config', 'control.yaml']
    )

    # Get URDF via xacro
    robot_description_command = [
            PathJoinSubstitution([FindExecutable(name='xacro')]),
            ' ',
            PathJoinSubstitution(
                [FindPackageShare('jackal_description'), 'urdf', 'jackal.urdf.xacro']
            ),
            ' ',
            'is_sim:=true',
            ' ',
            'gazebo_controllers:=',
            config_jackal_velocity_controller,
            ' ',
            'use_sim_time:=', use_sim_time,
        ]

    launch_jackal_description = IncludeLaunchDescription(
            PythonLaunchDescriptionSource(
                PathJoinSubstitution(
                    [FindPackageShare('jackal_description'),
                     'launch',
                     'description.launch.py']
                )
            ),
            launch_arguments=[('robot_description_command', robot_description_command)]
        )

    # Gazebo server
    gzserver = ExecuteProcess(
        cmd=['gzserver',
             '-s', 'libgazebo_ros_init.so',
             '-s', 'libgazebo_ros_factory.so',
             '--verbose',
             LaunchConfiguration('world_path')],
        output='screen',
    )

    # Gazebo client
    gzclient = ExecuteProcess(
        cmd=['gzclient'],
        output='screen',
        on_exit=[gzserver],
    )

    # Spawn robot
    spawn_robot = Node(
        package='gazebo_ros',
        executable='spawn_entity.py',
        name='spawn_jackal',
        arguments=['-entity',
                   'jackal',
                   '-topic',
                   'robot_description'],
        output='screen',
        parameters=[{'use_sim_time': use_sim_time}],
        remappings=[('/clock', '/clock')]
    )

    # Launch jackal_control/control.launch.py
    launch_jackal_control = IncludeLaunchDescription(
            PythonLaunchDescriptionSource(PathJoinSubstitution(
                [FindPackageShare('jackal_control'), 'launch', 'control.launch.py']
            )),
            launch_arguments=[('robot_description_command', robot_description_command),
                              ('is_sim', 'True'),
                              ('use_sim_time', use_sim_time)]
        )

    # Launch jackal_control/teleop_base.launch.py which is various ways to tele-op
    # the robot but does not include the joystick. Also, has a twist mux.
    launch_jackal_teleop_base = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(PathJoinSubstitution(
        [FindPackageShare('jackal_control'), 'launch', 'teleop_base.launch.py'])),
        launch_arguments=[('use_sim_time', use_sim_time)]
        )
        

    ld = LaunchDescription(ARGUMENTS)
    ld.add_action(gz_resource_path)
    ld.add_action(gzserver)
    ld.add_action(gzclient)
    ld.add_action(launch_jackal_description)
    ld.add_action(spawn_robot)
    ld.add_action(launch_jackal_control)
    ld.add_action(launch_jackal_teleop_base)

    return ld

closed 2025-01-30 by tonybaltovski